jukebox42/Octoprint-PrusaMMU #6: Future Considerations

jukebox42, 2022-08-01:
- Remember the previous tool so we can show unloading and loading at the same time with the two colors.
- Improve the loading message so it shows when the actual loading happens and not directly after (during multi-color prints)
- Handle print failures, these end up with the filament staying loaded so we should remember that and show it's still loaded.
- Move the mmu clearing code out of the js and into the python file, if we handle everything as part of the print then it should be always correct and we wont need the safety check.
- Give users a way in settings to clear out the mmu state
- Add datetime to the navbar title tag so hovering shows the user when the tool change happened (cause why not)

Feel free to comment below with features you'd like to see. Or open a pull request.

cvengels, 2022-08-05:
I'm always a fan of written names.
In this case, the color names.
Maybe in the future it's possible to merge these libraries into the project.
My own knowledge of JS is too bad atm :,)

[https://github.com/meodai/color-names](url)
[https://github.com/dtao/nearest-color](url)

jukebox42, 2022-08-23:
@cvengels I didn't go full color names yet but have added a better color picker with reasonable default values to save time.

closed 2023-03-18 by jukebox42