jukebox42/Octoprint-PrusaMMU #7: Spool Manager

naibaF5891, 2022-08-02:
Hi 
I love the idea and I see this as improvement over the PlugIn  "MMU2 Select Filament", but an Integration with existing spool managers would be extremely helpful. I for example am already managing my filament with the "Spool Manager" PlugIn.
Thank you and cheers
Fabian

jukebox42, 2022-08-02:
Good idea, let me add it to the list, if spool manager exposes a way to get that info I don't see why I couldn't add that integration.

naibaF5891, 2022-08-02:
You could check the "MMU2 Select Filament" Plugin for some input, there this is handled like this.

tonysurma, 2022-08-03:
+1 to above, I use the MMU2 Select Filament Plugin and it works with Filament Manager (what I use) as well for the filament names.

This plugin looks potentially more helpful than the select filament plugin so thanks in advance!

jukebox42, 2022-08-13:
Just published a version that adds support for both of these. I did some testing myself but I don't use either of them day to day so I wouldn't mind feedback :)

I also think I broke the version check in the last version so you might have to force an update(or download the latest below)

https://github.com/jukebox42/Octoprint-PrusaMMU/releases/download/2022.8.12/Octoprint-PrusaMmu.zip

schroding3r, 2022-08-15:
Seems like it's not quite working for me. When I begin a print, I get a popup asking if I'd like to use a given spool, but then it never seems to give that answer back to Octoprint. It waited for me to select a filament at the printer.

On a second test print, I set a default filament, and it waited for 30 seconds before selecting the default.
Please let me know if there are any other logs or anything I can provide you and thank you for a great plugin, it's so close to being _exactly_ what I needed.

jukebox42, 2022-08-15:
@schroding3r can you tell me what version you are running and if you are using SpoolManager or filamentmanager as your source? Was the title of the dialog you got Pruss MMU(as in did it look like the screenshot on the readme)?

schroding3r, 2022-08-16:
Ah I was running 8.12 somehow, but updated to 8.14 and disabled that reminder and now it gave me the proper spool selection pop up. It didn't give me the spool names or colors, just "1 2 3 4 5".

Spoolmanager does it's index starting from 0, so a checkbox to toggle start at 0/1 would be great too (if it's not too much to ask)!

Again, thank you so much. I'll continue to report and be happy to test.

jukebox42, 2022-08-17:
Interesting, So it should be displaying the names and colors from spoolmanager, if it's falling back to the numbers then it hit the failsafe(anything's better than nothing). Let me add a few more debug lines and share a version you can try. Appreciate the testing!

jukebox42, 2022-08-17:
@schroding3r I've added an alpha version that exposes some debug features available at this link:

https://github.com/jukebox42/Octoprint-PrusaMMU/pull/12

At the bottom of the Prusa MMU settings, you'll see a debug setting. Turning that on will cause Prusa MMU to write a bunch of debug information into the browser's console log. When you get a chance could you give the new version a try with debug on and send me a screenshot of the devtools console? It should look something like this:
[image]

schroding3r, 2022-08-17:
Is this what you're looking for? 
[image]
Is it ok if I just give you the screenshot instead of submitting a new issue? ;)

jukebox42, 2022-08-18:
@schroding3r perfect! Ok, new version available in the PR, please test when you get a sec :)

schroding3r, 2022-08-19:
Just gave that one an install and kicked a print off, looked great. Colors and names pulled. Thanks so much!

jukebox42, 2022-08-20:
thank you for the help! I'm going to resolve this issue now. I've released 2022.8.20 which includes the changes above. Don't forget to enable indexAtZero if you want your tools to report at 0-4 instead of 1-5(purely cosmetic) :)

https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2022.8.20

closed 2022-08-20 by jukebox42