jukebox42/Octoprint-PrusaMMU #8: Filament information from G-Code

cvengels, 2022-08-03:
Hi :)

When you export your print file you get lots of info about the used filament(s) in the comments at the end.
Maybe it would be possible to extract these and add them to the settings for each print (instead of manually setting the color and material)?

PrusaSlicer: 
[image]

First is the custom filament color you chose, second is the material color:
[image]

Basic material name:
[image]

Product name:
[image]

Thanks!

jukebox42, 2022-08-14:
Moving to under consideration. I like this idea but it's proven to be a bit too big for the last few releases.

jukebox42, 2023-10-07:
This feature is not currently planned so closing the open issue. Since the color data isnt available in single prints the user would still need to define it in the plugin, and changing the colors in an already sliced file isnt possible, so the data from the gcode would be fairly inaccurate the longer the file stayed on their octoprint.

closed 2023-10-07 by jukebox42

cvengels, 2023-10-07:
> Since the color data isnt available in single prints ...

It took me less than minute to find the filament color in the gcode file from a non MMU print.
Don't see any problem here why not to implement it.

[image]

jukebox42, 2023-10-07:
Sorry, you are right, in a single slice print a single color is there in the source. I noticed that too. But, it's only one color, sure it's there, but doesn't that defeat the purpose of having a modal to let you select what filament to use if it'll always assume it's that single color? Unless you plan to always print the same gcode with the exact same filament color, I don't think it'll be a very reliable feature.

As for changing the colors, yes, true you can edit the file. Gcode is easy to parse and read. But it's already been uploaded to octoprint. So each time you change your mmu colors, or want to print it in another color you could download the file, edit it, and reupload it. Or edit on the server through some plugin. But that's essentially what you can do today with the plugin without having to remember to update your files.

This feature could certainly be built. But I'm sorry, I don't see the value of trying to read the gcode data. It's more likely to be inaccurate the longer the gcode file exists in your octoprint unless you consistently keep the same colors in the same spots. Which is why I won't be prioritizing it.

Did not mean to spread misinformation. I'll add an addendum to my previous comment.