jukebox42/Octoprint-PrusaMMU #9: "Use Default Filament" command not executed and always disabled

sarf2k4, 2022-08-10:
I got this thing turned on hoping the whole thing could be 'automated' without any user input. I set the timeout above of this option to be 30 seconds, and enabled this. I started a print job, leaving the prompt to timeout by itself, then I still had to move over to my printer to select which filament I want to load for the printing.

When I checked the plugin setting by clicking via the navbar, "Use Default Filament" option became disabled.

jukebox42, 2022-08-10:
Oof, not good. Will figure out what's going on there.

jukebox42, 2022-08-13:
This issue has been fixed, it also looks like I broke the version checker so if it doesnt prompt you, you can download the latest here: https://github.com/jukebox42/Octoprint-PrusaMMU/releases/download/2022.8.12/Octoprint-PrusaMmu.zip

closed 2022-08-13 by jukebox42