jukebox42/Octoprint-PrusaMMU #13: incorrect update path?

jneilliii, 2022-08-17:
Someone mentioned an issue on the plugin repo https://github.com/OctoPrint/plugins.octoprint.org/issues/1101 and I think this may need to be addressed in your update hook here. 

https://github.com/jukebox42/Octoprint-PrusaMMU/blob/d1b358541eef969ae8e5139a3fa1f9622823360c/octoprint_prusammu/__init__.py#L403

jukebox42, 2022-08-17:
Well that's embarrassing. Thank you for reporting it. I just published a  new version that fixes the issue.

@jneilliii I assume for everything to fix their install they'll need to re-download the plugin?

jneilliii, 2022-08-17:
They can uninstall (not erasing the data) and then reinstall from plugin manager and it should automatically correct it.

closed 2022-08-20 by jukebox42