jukebox42/Octoprint-PrusaMMU #21: UI shows "Unknown filament" when using Spool Manager after reload

40thoughts, 2023-02-06:
Here's a little something that's bothered me from some time, not that it has any functional impact :

jukebox42, 2023-02-07:
Ah, yeah. This is a race condition issue I should have probably fixed by now. The short of it is, when the UI loads the spool manager DB hasn't initialized by the time we ask it for the filament info, so it falls back to the unknown state. I should be able to get around it, just haven't spent time debugging it

jukebox42, 2023-02-20:
Hey @40thoughts,

Would you be willing to test a fix for this? I don't regularly use a spool manager. I've done some initial testing but would love if someone who did use one could give it a try.

Package to install is available here: https://github.com/jukebox42/Octoprint-PrusaMMU/pull/22

40thoughts, 2023-02-26:
Hey @jukebox42, actually I installed some other plugins and it apparently didn't happen anymore, I tried reloading the page a lot on my last prints and it always showed the right filament type.
But this morning it happened again (from another pc) so I'll try your fix on my next print, and let you know.

jukebox42, 2023-02-26:
Thank you @40thoughts. I just released this new version so you should be able to just download the latest release now. I ran a few more tests and I'm fairly confident it's fixed now. I'm going to close this issue but if you run into any issues please let me know :).

closed 2023-02-26 by jukebox42

40thoughts, 2023-03-01:
Hey @jukebox42, I updated your plugin and it seems to work just fine. Great job!

However, I noticed another thing and didn't know if it deserves another ticket so here it is :
When printing in single mode (still using Spool manager), your plugin's giving us the possibility to choose between different spools, but after choosing, the UI still shows "Tool 0" as selected and if there isn't enough filament on Tool 0, Spool manager is giving a warning that there isn't enough material to get it done.
I'm not taking the warning into account since it still works fine and seems to decrement the right spool at the end but showing the right spool info would be super nice.

[image]

I believe debugging that would be a real mess between plugins, I'll just let you decide if you'd like to open an issue for that by yourself since I wouldn't like to tackle it myself...

Still, your plugin is awesome, I'll be glad to overlook those nasty little details, just figured you'd like to know about it.
Again, great job! Thanks for what you've done for the community.