jukebox42/Octoprint-PrusaMMU #36: !!! Before you report !!!

jukebox42, 2023-09-27:
Things to check:
- Make sure you are running the latest version of Prusa MMU
- Make sure your printer profile has "Number of extruders" set to `5` AND "Shared nozzle" is checked.
- Make sure you are slicing your prints with PrusaSlicer and using a printer profile that supports either MMU or MMU Single.
- Make sure your Octoprint version is greater than or equal to `1.8.0` (when they dropped python 2 support)

Needed to debug:
- What version of the Prusa MMU plugin are you running?
- What versions is your printer running? (and what version is your MMU running?)
- What behavior are you seeing?

Known bugs:
- The nav menu can sometimes get stuck if the print finishes before you return to the tab (working on a fix)
- The nav menu may report an incorrect state on occasion. Sometimes the MMU reports a state message in the wrong order.

ggppjj, 2025-01-20:
Apologies, this isn't the right place to ask this question that has nothing to do with your plugin but it's oddly difficult to find others with Octoprint/MMU3 enabled. Do you have yours on an MK3S+? If so, do your filament load test squares ever clear from the LCD after it loads filament during a print started from Octoprint?

I'm seeing this issue and it's such a niche issue in such a niche combination of hardware (apparently) that basically nobody else is reporting this anywhere I can see, which is part of why I'm asking here.

jukebox42, 2025-01-26:
@ggppjj Is this when it loads the filament and checks, that it draws the boxes on your LCD and some of those boxes are blank instead of filled in? That means that the filament didn't load properly, usually means you need to fix the alignment by your extruder or the filament head was too wide to fit into the hole where the sensor is.

ggppjj, 2025-01-26:
My issue was caused by being on the dev branch of the DisplayLayerProgress plugin, it changed out all `M117` commands with `M118`, which the MK3S+ doesn't display on screen. Thank you for replying, apologies for not following back up.