jukebox42/Octoprint-PrusaMMU #40: State hangs on Connecting

dschnur, 2023-10-25:
Error loading with MMU3 / I3Mk3s+
Printer Firmware: 13.31
MMU3 Firmware:  3.3.0
Raspberry Pi Zero 2w
RPI OS: Raspbian GNU/Linux 10 (buster) (Packages up to date)

Symptoms:
Octoprint connects via serial, but State: hangs on "Connecting"

Work Around:
  Edited: 
  Commented out lines 580 -> 591 
  /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusammu/__init__.py

Log below:
2023-10-25 14:17:31,789 - octoprint.plugin - ERROR - Error while calling plugin prusammu
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_prusammu/__init__.py", line 453, in on_event
    self._log("on_event {} with".format(event), obj=payload, debug=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_prusammu/__init__.py", line 580, in _log
    if self.config[SettingsKeys.DEBUG]:
KeyError: 'debug'

jukebox42, 2023-10-28:
Huh, curious. I wonder why debug isn't in your config. I'll take a look at this, this weekend. may be a mistype somewhere or I need to add some negative checks.

jukebox42, 2023-10-28:
Looks like I had an order problem where log was being called before settings initialized. I've added some checks around it to fix that issue and over-engineered the safety checking there to avoid this in the future. Should be fixed now, I didnt see any log errors when I was testing the new one. 

https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2023.10.28

Please let me know if you see it still.

closed 2023-10-28 by jukebox42

dschnur, 2023-10-29:
Wow, that was quick!

I’ll test it this weekend and tell you what I see.

Keep up the good work!

 -Dan

dschnur, 2023-10-29:
So far so good.  I did quick test print with one color and it went without a hitch.

I’ll try one with a few colors and I’ll cause a feed failure just to test how that situation gets resolved tomorrow.

Thanks for the help!

 -Dan