jukebox42/Octoprint-PrusaMMU #41: Error pop up spam

Madcamper, 2023-10-26:
Not sure where else to put this but is there anyway to disable the error pop up? Since the MMU retries multiple times on a failure to feed these error boxes take up the entire screen and the only way to get rid of them is click the X on each error.
[image]

jukebox42, 2023-10-28:
Oof, not good. yeah Leme add a setting for it. I may roll that change back until i can come up with a more stable solution.

jukebox42, 2023-10-28:
I've disabled the popup in the latest release. I'll bring it back as soon as i can work around the spam issue. Data is still available on the settings page when it errors out.

https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2023.10.28

Kevman323, 2023-10-28:
I don't believe the spamming is a bug with this plugin's code, well, not exactly. Since the 3.x.x firmware auto retries, it might throw up to 3 errors, and if it succeeds on/before it's 3rd retry, the print will continue. If you get small subtle issues that fix themselves, they can pile up over the course of a print. This plugin already ignores errors during the auto retry phase, and only changes the state to "ATTENTION" when it detects another line signifying that the printer is waiting for a user.

As a quick fix, perhaps you could make sure the current state is "ATTENTION" before throwing the error, that way the error box will only pop up when the printer is paused and waiting for the user. It still might throw another error box if the user encounters the error again while trying to manually fix it, but presumably it won't make too many boxes, and the user would be there to close the few boxes that appeared.

Ideally, if there's a way for the plugin to clear the error messages it has created, perhaps you could automatically clear them once the state returns to "OK", but knowing that a background error that was fixed with an auto-retry could be useful. Alternatively, maybe you could only create one error message, and check if it already exists, then edit it with the new information instead of creating multiple. Not sure if the Octoprint API allows for this though.

jukebox42, 2023-10-29:
Yeah, it's actually working as intended, it's showing you the errors as we read them, but the spamming is less ideal.

Madcamper, 2023-10-29:
I don't remember exactly. iirc, it pops one up each time it fails to feed and the MMU I think tries 3 times before requiring manually intervention which it also throws a pop up for.

jukebox42, 2023-10-29:
No worries. I'll see if I can recreate the issue should be fairly easy to replicate.

closed 2023-10-29 by jukebox42

jukebox42, 2023-10-29:
There's likely a bit more I could do to improve it. I took an arrpoach to replacing the error on the popup if we get one in while it's being shown.

If this proves to still be an issue I'll probably take @Kevman323's approach and only show the error when we send ATTENTION. Please let me know if you run into this issue again. New version `2023.10.29` should be available once octoprint syncs.