jukebox42/Octoprint-PrusaMMU #46: Support for Spoolman

SirFlip, 2024-05-15:
Support for the Spoolman plugin https://github.com/mdziekon/octoprint-spoolman would be a great addition.

Then it would nicely work together with https://github.com/Donkie/Spoolman

jukebox42, 2024-05-18:
I'll take a look once I get this MK4 version stable. Hopefully they have similar hooks to the other spool managers I've hooked up.

GitSoks, 2024-08-24:
+1 Support for Spoolman would be really awesome

crzykidd, 2024-09-01:
+1 for Spoolman, Thanks for all your effort.

Llamrei80, 2024-10-04:
Any Update on this?

Great plugin, keep up the good work.

jukebox42, 2024-11-02:
Hey All,

I appreciate your patience. Finally got around to poking at this. I didn't realize I was signing myself up for spinning up another service lol. Took me a little bit longer  to figure out how to capture the data (I really need to do some refactoring some day).

I'm not a regular user of spool man so I was hoping for a few testers to let me know if things are working as expected. Here is a patch version of the plugin with spool man support :).

A few notes:
- You'll need to go into the prusa mmu settings and switch over to SpoolMan
- Filament described as multi-color in spool man will show up as one color in the filament window/nav icon (the first one defined)

To install:
- Download the zip file below (or build this pr #55 yourself and use that zip file)
- Within Octoprint navigate to settings -> Plugin Manager
- Click `+ Get More`
- Enter your password
- Scroll to the bottom and upload the file
- Restart.

[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/17608638/Octoprint-PrusaMmu.zip)

Thanks in advance!

DavyGD, 2024-11-04:
Giving it a try!

graelo, 2024-11-13:
Hi @jukebox42, thank you so much for this (and more generally the plugin).

I just successfully printed these 3 configs, and each time your plugin updated spoolman using the precise weights from the gcode:

- single tool print with tool 0
- single tool print with tool 1
- multi-material (just one tool change because testing) with both tool 0 and 1

Here are the screenshots for the 2 first configs

<details>
with tool 0

[image]

</details>

<details>
with tool 1

[image]

</details>

I have 2 remarks, but honestly not sure they are related the Spoolman integration

- my initial print was my first one with your plugin, and I mistakenly selected a tool override -> this led to a cryptic error about T8 not being passed to the printer. Probably rings a bell to you.
- the other 3 prints were all successful, with the expected filament, but the filament shown in the menu bar is not at all the one being selected for printing. This is just a display issue.

HIH!

graelo, 2024-11-18:
Hello, additional pieces of feedback. 

Great:
- After many multi-material and single material prints, no issues with printing / filament choices

The following are minor issues. In my view, there's no need to address them as part of this PR: they could be the topic of other issues. As you see fit.

Minor:
- the filament shown in the menu bar is quite never aligned with my choices in the spoolman plugin: meaning if tool #1 is gray in the spoolman plugin, the MMU menubar filament would display, say, Orange. This is visible in my first screenshot above.
- when a print starts, the plugin displays a summary window (very useful), but in case Tool #0 was not used, the window would still present it as using 0.0g.

Less minor:
- the prusammu plugin updates the date of all filaments, even if some of them were not used in the print. Because all filaments are updated (with 0.0g), all update dates are refreshed, and it much lessens the usefulness of the Last Used Date in spoolman (and I guess in other plugins too, as you made no change in your spoolman branch).

I hope it helps. Thanks again!

graelo, 2024-12-16:
Hi, thanks @jukebox42 

I can't remember what was my config at that time. But what I'm noticing in my current print could help.

- I'm printing with the 3rd tool (Tool 2)
- the color and material displayed in menu bar ("Gray (PETG)") do not correspond to any color in my current swatch.
- the number does not correspond to the current tool used either (or it's an +1 index ⇧)
- I did use Gray (PETG) in the past, in 5th position (Tool 4), but I since changed the spool to some other material.

<details>
[image]
</details>

jukebox42, 2024-12-22:
Huh, that's interesting. It's possible I'm not reading the right placement ids when displaying the filament. I'll do some digging

jukebox42, 2024-12-22:
Alright, turns out I was pulling just the raw spool list and not the active selected list. So it was sorting by index and wasn't taking into account if it was actually being used. TL;DR: I was doing it wrong.

Here's a new build if anyone could test when you have time.

Thanks in advance!
[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/18222669/Octoprint-PrusaMmu.zip)

jukebox42, 2024-12-22:
Thank you for testing! I just merged and released this as my latest version. The release is identical to the version you tested, just with an official version.

I'm going to resolve this thread. Please open a new bug if you find any issues with it.

closed 2024-12-22 by jukebox42