jukebox42/Octoprint-PrusaMMU #50: MK4: Only show the filament prompt if the file was sliced with one filament

jukebox42, 2024-08-15:
Opening this issue now since I know it's a sore spot.

Currently for MK4 the prompt will show for all prints. To fix this I need to parse the gcode comments and look for the number of tools being used.

MysticGringo, 2024-12-14:
I wonder if a solution is to have people give a particular name to the Print Settings Profile, one that your plugin looks for in order to issue the prompt? This would show up in the "print_Settings_ID".

jukebox42, 2024-12-15:
I've thought about a couple options. One I considered was reading the "tool" list in the gcode file and checking to see how many they have, if it's more than one treat it like a multi color print, otherwise don't provide an override.

ruedli, 2025-02-17:
The MK4 / MMU3 supports "tool remapping" in a menu of the MK4, before a print starts.

On octoprint this is not the case as far as I can see.

MK4 buddy board support M863 - tool mapping GCODE, see https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112

In short:

**M863 - Tool remapping**
Allows to redefine which tools use for certain parts of the print. 
If a part of an object is sliced to be printed with filament 1 - but you wish to print it with filament 2 instead, this G-code helps you achieve so.

Examples:

M863 M P0 L1 : Instead of tool 0, use tool 1
M863 E1/0 : Enable/disable tool remapping
M863 R : Reset tool remapping
M863 : Report current tool mapping

M864 - Spool join settings

A menu to define such remapping (original tools, with the tool to select in a pulldown), would then issue appropriate M863 commands (starting with a reset and then setting them) as part of the startup commands.

Note that a similar support could be used to define spooljoin features, see M864. Defining spooljoins other then the default "next spool" makes sense for multicolor prints.

M864 - Spool join settings
Enables to configure Spool join function, defining which material to use after the one used runs out.

Examples:

M864 J A1 B2 : When tool 1 runs out of filament, continue with tool 2
M864 R : reset any settings
M864 : Report current spool join settings

jukebox42, 2025-02-23:
Hey @ruedli,

There's a few threads where we tried this back when we were trying to implement MK4 support. The biggest issue with tool remapping is you can't map to the same tool more than once, trying to do so results in errors, and mapping to something that was disabled throws some pretty gnarly errors. That was my initial plan for supporting the single print mode. I ultimately settled on just remapping via the plugin because of this.