jukebox42/Octoprint-PrusaMMU #51: Question concerning Update 2024.8.15

KayGundhardt, 2024-08-22:
Thank you very much for your work and effort on the update!!! 

:)

In your release notes you write that when printing with only one color using the MMU profile, the slicer setting is overwritten by the tool selection. Does this mean that the last selection made for a tool before printing also overwrites the tool selection selected and sliced in the MMU profile afterwards?

For example, I manually select Tool2 (e.g. black) via the plugin, print something and then switch off.

A few days later I slice something, select Tool3 (e.g. red) with the MMU profile and start printing directly without setting something via the plugin.

Is the model then printed in black or red?

jukebox42, 2024-08-23:
Hey @KayGundhardt:
It depends on what version Prusa you are running.

**MK3/MK3s**
If you are using an MK3 everything will work as it normally has, no change to your experience, the latest release is directed at MK3.5/3.9/4 users. Slicing with MMU Single profile will result in a prompt at the start of a print to ask you what filament to use. Slicing with Multi mode will result in no prompt and will print as it normally has.

**MK3.5/3.9/4**
If you are using an MK3.5/3.9/4 then there's a slight behavior change:
Because there's no "MMU Single" profile slicing with multi and starting a print will always prompt you with the filament selection option. Selecting a filament will cause all tool changes to be interpreted as your selected tool. There's also a "skip" option in the prompt which will cause it to print as it was sliced. No settings are preserved across prints so every time you start a print you will be given the same prompt to pick a filament to print with.

KayGundhardt, 2024-08-25:
Okay, as I have a MK3S+ I seem to be lucky.
:))

But for the others (MK4 and others): The situation you described is only relevant if you slice using the „standard tool“ and not specifically e.g. Tool3, right?

jukebox42, 2024-08-31:
Yeah, so for the mk3 there's a profile called MMU Single. When you slice
with it, no tool is selected by default and when you print with it (without
my plugin) the printer screen will ask you to select a tool. With this
plugin the tool selection prompt is inside octoprint.

If you slice with the MMU profile, you select the tool during slicing and
it will print that as normal.

KayGundhardt, 2024-08-31:
Thanks a lot for making this clear :)

cj8scrambler, 2024-09-02:
What is used to make the decision of which printer is being used?  I have a MK3S + MMU2S and I'm using "MK3S+MMU2 Single" profile in PrusaSlicer.  I thought the behavior would remain unchanged for me, but it's not.  When I send a print, I'm prompted to choose a filament from the popup and I do, however after chosen, the printer LCD is still prompting for a filament choice.  It won't proceed until I select a filament there on the printer.

If I use the multi-filament profile in PrusaSlicer and assign a filament, then when I send the print, I'm prompted for a filament from the popup and once chosen everything works fine.  I haven't tried choosing a different filament in the popup from the one assigned in the PrusaSlicer to see which one it actually uses.

KayGundhardt, 2024-09-02:
So this is a bug then, right? Or even two bugs? 

When I previously understood everything right using the single profile, octoprint should prompt you and mk3 should accept. Second: multifilament profile should prompt you nothing at all.

Only difference is: I asked for an MK3S+ with MMU3…

jukebox42, 2024-09-02:
Oof, that sounds like a bug. I'll take a look at that tonight. It shouldn't
be prompting you twice.

I'm using the machine message to detect the print but maybe there are other
variations I wasn't anticipating.

When you connect your printer, can you look for the long startup line from
the printer that starts with FIRMWARE_NAME and share that with me? Feel
free to remove your uuid from the long I just need the other bits.

Thanks!

cj8scrambler, 2024-09-03:
> Oof, that sounds like a bug. I'll take a look at that tonight. It shouldn't be prompting you twice. I'm using the machine message to detect the print but maybe there are other variations I wasn't [...]

Apparently I don't have a UUID to hide (this is the actual output):
```
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.11.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
```

jukebox42, 2024-09-03:
Thanks @cj8scrambler!

I have the same blank uuid issue, I think that's just an MK3 thing. I'm glad you get that post. I think I have a workaround for this issue. Work is in this PR #53. I plan to test it as soon as my current print finishes and once I'm happy with it I'll ask for some testers.

If anyone with an MK3 wants to work around this issue, you can manually install this release https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2023.10.29

jukebox42, 2024-09-03:
Alright, I have a new version I'd love for you all to try. This simplifies the prusa detection, I tried to get too fancy. This should hopefully get around the missing machine type detector. I've also defaulted more to the MK3 version so even if it fails the detection it should force the MK3 version.

Source for this build can be found here #53 

[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/16843299/Octoprint-PrusaMmu.zip)

cj8scrambler, 2024-09-04:
Tested out the update and this fixes my issue on MK3S/MMU2S.

KayGundhardt, 2024-09-05:
Okay, seems af it’s working now - thanks for the fast reply and patch :)

closed 2024-09-05 by KayGundhardt