jukebox42/Octoprint-PrusaMMU #52: MK4s + MMU3 reporting unknown MMU status

Java-SE-Binary, 2024-08-23:
Printer model: MK4s + MMU3

PrusaMMU plugin version: 2024.8.15

Printer + MMU version: 6.1.0 + 3.0.3

Behavior: The plugin does not recognize the current state of the MMU, whether it be idle, loading filament or otherwise.

jukebox42, 2024-08-23:
Hm, I must have broken that when I fixed support for filament selection. I'll take a look at the diff and see where I might have broken it, thanks for the report.

jukebox42, 2024-08-25:
Alright, had a chance to look into this. I hopefully found the problem but since I've only got an MK3s I can only test with the virtual printer.

Would you or anyone watching this thread be able to give this version a test? It would be ideal to test with a multi-filament print so you can watch the load/unload. You should see it say it's OK when the printer connects, show loading a filament near the start, changing between two when it swaps, and then finally unloading at the end.

For debugging, the plugin has a debug mode on the settings page under advanced options. This will cause it to print out messages into the log as well as your browsers devtools. (once done remember to turn off as it can slow performance)

Thank you in advance!

[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/16741815/Octoprint-PrusaMmu.zip)

Java-SE-Binary, 2024-08-25:
Had a chance to test it and enable debugging. The fix did not change anything on my end. No matter what's done, the navbar still reports unknown.
Attached is a screenshot of the relevant lines in the browser console.

[image]

jukebox42, 2024-08-26:
Hmm, @Java-SE-Binary can you try disconnecting and reconnecting (using the button in octoprint)? I'd like to see the firmware trigger. When you connect, can you look for a long line that starts with `firmware_info_hook:`? It _should_ have said `MK4` in `prusaVersion` but yours is saying `null`. Feel free to black out the UUID in the response, I don't need it :)

Can you also look for a line that looks like `{action: 'debug', msg: 'gcode_received_hook MMU: M862.6 P "MMU3"'}`. Both of these should show up during the printer connection process.

Thanks in advance!

Java-SE-Binary, 2024-08-26:
Firmware hook here.
[image]

Unfortunately, I'm not seeing a line containing a gcode_received_hook, and as a matter of fact, I'm not seeing any info regarding my MMU in the gcode terminal like I used to on my MK3s. I set up this MK4s somewhat recently, so I'm still getting used to its quirks, and all the differences that come with the xBuddy boards/firmwares.

jukebox42, 2024-09-02:
@Java-SE-Binary Oh that is interesting. Have you verified you've installed the MK4 with MMU firmware instead of the base firmware?

Java-SE-Binary, 2024-09-03:
There is no difference between the firmware that is released for the MK4's and the MK4 firmware that comes bundled in the ZIP file under the MMU sections of firmware downloads. This was the same case for the MK3's as well, as I've had experience compiling my own firmware for the older printers. This can be seen as the printer firmware bundled with MMU firmware does not have special naming to it. I did flash the firmware from the MMU bundle just to make sure, and there was no change in behavior. 

I believe the firmware is supposed to start reporting the presence of an MMU when the MMU toggle switch is enabled in firmware, for new and old printers.

jukebox42, 2024-09-03:
Hm, TMYK.

I have a new build that might work. Some folks are having issues with the machine type detection so I've tried a simpler method of detection. I've also ditched the M862.6 check, it wasn't doing much since I assume everyone using this plugin is using it with an MMU.

If you have a sec please let me know if it works for you. Thanks!
[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/16843347/Octoprint-PrusaMmu.zip)

Java-SE-Binary, 2024-09-04:
Gave the new version a shot, didn't get very far unfortunately. The plugin is recognizing my MK4S as a MK3, so it seems things aren't working quite as expected.

[image]

jukebox42, 2024-09-08:
This is peculiar. It's interesting it's not noticing the MK4 version even though it's in your firmware command.

I have a more brute-force approach I can take. I can move the version to a dropdown and let you just hard-code it. I'll see if I can get you a version this weekend for you to play with and see if it fixes your issue. You might be the first MK4s to try this plugin.

jukebox42, 2024-09-09:
I've added an override setting in my latest release :) https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2024.9.8_1

Java-SE-Binary, 2024-09-09:
Looks like this update has everything working beautifully! It successfully hooked from the M115 command and is displaying MMU status in the navbar! Only minor hiccup is Tool 4 (5th slot on MMU) does not switch from a 'Loading to Nozzle' state to a 'Loaded' state. Other tools display just fine, so its not a super pressing issue!

Great work and thank you so much for working on this issue! :)

jukebox42, 2024-09-14:
That's interesting. I wonder if that was a fluke (sometimes it misses the indicator) or a bug I have. Let me know if you see that again and I can do some additional digging. Going to close this for now since it looks like your main issue is fixed :)

Appreciate the help debugging!