jukebox42/Octoprint-PrusaMMU #56: Spool Join feature question?

collateral87, 2024-11-09:
Hi,
i have a question about the spool join feature of the mmu unit.
Is there any option or advice to enable this or something similar with this plugin. I want that the printer changes from tool 1 to tool 2 when the filament from tool 1 is emty for example. 

ps: thank you for your genius work, to make the mmu usable in octoprint
pps: and thank you to implement spoolman ;)

jukebox42, 2024-11-12:
Hey @collateral87,

Currently, there's no way to enable/disable spool join from the plugin. I'm not sure if that feature is exposed or if it's only available internally, but I can look. You can, however, set spool join on your printer, and the plugin will work with it without a problem. Just make sure you don't have any overrides set in the plugin (under advanced) since those may contradict the spool join setting.

collateral87, 2024-11-12:
Thank you for your reply. The problem is, that i can select spool join only, when i start a printjob from the usb. Then a popup comes up where i can define the spools and select spool join.
When i print with octoprint, there is no popup. Or i miss something?