jukebox42/Octoprint-PrusaMMU #57: MMU with more than 5 filaments?

MysticGringo, 2024-12-13:
Hey Jukebox...

Your plugin has been working fantastically... but I've made a 12 filament MMU (https://github.com/cjbaar/prusa-mmu-12x) and wondered how to tell your plugin that I have 12 filaments now? I am sure it's not a high priority, but it would be cool if there was a way to select how many filaments I have loaded.

Thanks!

jukebox42, 2024-12-15:
Hey @MysticGringo, it's not currently possible out of the box. It would require a bit of refactoring since a few things are hard-coded expecting 5 filament slots.

I'd be willing to accept this as a contribution if someone wanted to take a stab at implementing it but it's not on my list of things to support. It would work best if it was a new advanced option to provide the number of filament slots available. Testing it with the multiple spool managers is likely the hardest part.

MysticGringo, 2024-12-15:
Thanks for the response... Yeah, I figured it was a long shot but worth asking. I am sure this is a relatively rare case that not many people will even need.

jukebox42, 2025-01-05:
Hey @MysticGringo,

So I have no way to test this but since you've been so helpful with my other releases I figured I'd take a stab at supporting more than 5.

I ran some virtual tests and it seems to be ok but I'd appreciate it if you could confirm the selection works and the filament changing works past 5 :).

To use:
- Install the attached version
- Open settings, then expand advanced and look for "Filament Count"
- Set that number to 12 (or whatever)
- Save it
- Reopen and then you should see more options on the filament list.

Let me know how it goes! The PR is here if you're curious #58 
[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/18311979/Octoprint-PrusaMmu.zip)

MysticGringo, 2025-01-05:
Awesome! Thank you. I’m out of the country right now, so I’ll test it when I get back to the printer next week. 

Thanks again!

3d-gussner, 2025-01-08:
I tested the provided ZIP above with the MK404 SIM (modified for the MMU 12x) using my [FW 3.14.1 for MMU 12x](https://github.com/3d-gussner/Prusa-Firmware/releases/tag/c3.14.1_MMU12x) and @cjbaar [MMU3.0.3_12x firmware](https://github.com/cjbaar/prusa-mmu-12x/blob/master/firmware/cjb_fw_mmu3_3.0.3-x12.hex).

Octoprint 1.10.3 Python 3.10.12 and Printer profile with `Number of Extruders: 12`

- [X]  Setting 12 tools in Advanced option :heavy_check_mark: 
- [X] `T0` to `T9` or `Load to nozzle` `1` to `10` works well :heavy_check_mark:

[image]

- [ ] `T10` and `T11` or `Load to nozzle` `11` and `12` doesn't show correctly. `NaN Unkown Filament` :warning:

[image]

Hope this helps.
BTW: I still don't own the MMU12x just trying to help.

jukebox42, 2025-01-09:
I just learned there's an MK4 SIM, woulda helped when I was doing the MK4 integration lol. Thanks, this helps I'll take a look at what's up with the NaN for those. I think I might know where I forgot to account for the extra digit. Appreciate it!

MysticGringo, 2025-01-12:
Hey Jukebox...

I've just tried it with a single filament in position 5, and everything worked as expected. I'm building a new filament mounting shelf for the 12 filaments, but will setup a test over the next few days and will test some of the other filaments. So far everything looks like it works correctly.

Thank you for this, I really appreciate it, and hopefully it pays off if there's ever a new official MMU with more filaments.

jukebox42, 2025-01-26:
Thanks for the additional testing @3d-gussner.

Here's a new version that should fix the problem you saw.

[Octoprint-PrusaMmu.zip](https://github.com/user-attachments/files/18551422/Octoprint-PrusaMmu.zip)

jukebox42, 2025-01-27:
Thanks for all the testing! I've merged this pr and released it. I just realized i forgot to update the credit list for your help. I'll get that done this week. Appreciate all the help!

closed 2025-01-27 by jukebox42