jukebox42/Octoprint-PrusaMMU #59: No longer able to change colors in dialog boxes

Evilmonkeybot, 2025-01-23:
Until recently, the plugin worked as expected. Now, however, I am unable to change the color of each filament via the drop-down beside the filament label.

Original Mk3.5, MMU3, Octoprint 1.10.3, PrusaSlicer-2.9.0. Latest firmware on Mk3.5 (6.1.3+7898) and MMU3 (6.1.3 (3.0.3)) - no changes for several weeks. PrusaSlicer update was recent.

[image]

jukebox42, 2025-01-26:
Hey @Evilmonkeybot, 

What browser and version are you using? That would be a js bug and likely not related to Prusa.

Additionally, did you install any plugins recently? It's possible there's two plugins causing a js error.

If you are familiar with devtools, could you open up your browser's developer tools, click on "console" and see if you have any errors showing? Would help me understand where the potential issue is. Thanks!

Evilmonkeybot, 2025-01-26:
Both PrusaSlicer 2.9.0 (for Mac arm64) "Physical Printer" tab, Chrome 131.0.6778.265 and Safari 18.2. The plugins that were installed and active before the glitch were OctoEverywhere and Cancel Objects.

The console error on the page is attached below.

[Octoprint Console Errors.log](https://github.com/user-attachments/files/18551303/Octoprint.Console.Errors.log)

jukebox42, 2025-01-26:
Perfect, This tells me where to look, thanks. This should be an easy fix.

Evilmonkeybot, 2025-01-26:
I just deleted and re-installed the plugin. Now I notice that the status had also changed and was showing xUnknown all the time. With re-install, the status shows Idle appropriately and the filament color dialog pops up as it should. Not sure what broke it, but for now the re-install has fixed it. I will have to see if that lasts through a print cycle.

jukebox42, 2025-01-26:
Sounds good, the re-install should fix it. I have a fix for it already I'm pretty sure. I can't seem to replicate the bug, but this will prevent the unknown color problem from happening.

#60 I'll get it rolled into my next release

closed 2025-01-27 by jukebox42