jukebox42/Octoprint-PrusaMMU #61: Does Mk3s+ selection no longer work as with the Mk3?

DracoZaroff, 2025-02-01:
I recently had to remake my octoprint server. Prior to this the single filament selection worked nearly every time with a few minor exceptions. Now after reinstalling everything it seems the filament selection doesn't work and requires me to select it on the printer itself. 
My printer is the Prusa Mk3s+ with MMU2S. Both are running the newest firmware. And my Octoprint server is running the newest version as well.

jukebox42, 2025-02-02:
There's a few things that could cause it:
- Can you verify the plugin is detecting your printer correctly? Once you connect, if you open settings you should see the printer detected in the bottom (might be under advanced). If it's wrong, try forcing it to MK3.
- Next time you try to print can you check if your printer complains about the "gcode sliced for a different printer"? This will stall the color modal because the printer has to do something, then by the time it dismisses the signal for the color picker (temp) gets passed. This is fixed by restarting the printer (weird mk3 bug) or using different gcode
- Do you have a custom startup script in your gcode or in octoprint? If you've altered the base one it's possible the right command isn't in there anymore or has been altered (common if you alter the gcode to do a pre-heat)
- Do you have another plugin that is also doing something on startup? the timer for the popup is pretty short so it may get missed if something else has popped up.

Let me know if that helps, if not please turn on debug mode, open up your browsers "console" (right click, developer tools, console), start a print, and send me a screenshot or export the log. I can take a look and see if something else is not working.

DracoZaroff, 2025-02-04:
- It does detect everything correctly and I have even tried to force it to MK3 to be sure. That doesn't seem to have any effect on it. 

- Since my last firmware update I do get the gcode sliced for a different printer message. But it had still been working even with this behavior before I had to remake my octoprint server so I didn't consider it as a possibility. (If you have any insight into how to correct that I'd love to resolve it. It's just been lower on my priority list as it's more of an inconvenience I expect to be fixed with a future firmware update.)

- I can't recall ever having altered the starting script through slicer. If I had it was a long time ago and shouldn't be any different from when it had worked last.

- When installing this time I noticed there was a list of recommended plugins to preinstall. They all seemed pretty benign so I opted to install all of them. They came from the installation script I used "https://github.com/paukstelis/octoprint_deploy". The list can be found at https://github.com/paukstelis/octoprint_deploy/blob/master/plugins_list

[image]
[192.168.86.8-1738631489251.log](https://github.com/user-attachments/files/18650250/192.168.86.8-1738631489251.log)

DracoZaroff, 2025-02-12:
I'm still stumped as to what to do to fix this. If there's any additional information I can provide that might help fix this, let me know.

jukebox42, 2025-02-23:
Hey @DracoZaroff,

Sorry, works been crazy so I havent had a chance to get back to this.

Out of that list you've installed these are the ones I havent tested with:
- CancelObject 
- Bed Visualizer
- Terminal Commands Extended
- Active Filters Extended
- PowerFailure 

The two that could potentially be causing an issue are: CancelObject and Active Filters Extended. Both seems to hijack gcode and it's possible they are catching or silencing the posts. I'll try installing all of these today and see if i can get it to fail.

That Forbidden error you are getting there is concerning. When you re-set up your octoprint did you change any of your user permissions? Settings -> Access Control. I believe you'll need at least Operator access.

Something else to try, you could try downgrading to this version of the plugin. The most recent release introduced some dynamic features for MMUs with more than 5 options and I worry I introduced a breaking change for first time users, something I can't really test. https://github.com/jukebox42/Octoprint-PrusaMMU/releases/tag/2024.12.22

jukebox42, 2025-02-23:
Alright, I just ran a test with all of the plugins above and I still got the prompt when I started a print and it selected correctly. I did leave them at their base configs. Do you know if you made any changes to those plugins when you set them up?

I think we can rule out any start script changes if you cannot remember, the base detection hasn't changed since I first wrote this plugin.

Looking at your screenshot though, I think it's permissions related. Octoprint rejected the filament selection call.

Can you try this:
- Start a print
- Make a firmware selection
- Let it error out
- stop the print
- Then go get me a copy of your octoprint.log file

Feel free to scrub anything you want. I'd like to see if there's any python errors in the log and if they relate to this plugin. Could help diagnose the issue.

jukebox42, 2025-08-10:
Closing due to inactivity, reopen if you still need help.

closed 2025-08-10 by jukebox42