jukebox42/Octoprint-PrusaMMU #63: Non-admin user filament management

chrissexton, 2025-02-26:
I don't see it as a particular feature, but upon attempting to allow my students to manage the filament colors/enabled/disabled status, I discovered that users with Operator status can see the MMU settings page, but not modify it. That in itself doesn't surprise me, but I would like to allow them access to change the spool settings as they are allowed to load and unload filament from our machine. I don't see an available permission for this. Is it possible?

jukebox42, 2025-02-27:
Alright, so I did some digging. Unfortunately, I don't think it's possible without granting them permission to manage all settings. Permissions in Octoprint aren't very granular and the plugin stores the spool properties in the settings store, I don't have control over what permissions are evaluated when writing to settings.

I think there's two options:
1. You add Settings Management to the Operator permission (you can do that under User Access)
2. You set up a third-party tool for managing filament outside of Octoprint and let them manage the spools that way. I know SpoolMan would support your use case, though it requires you to set up a third-party UI. You could also check FilamentManager and SpoolManager, one of them if I remember correctly uses an sqllite database instead of settings so it may not require elevated permissions to save. Installing any of these plugins will let you point the filament list to those plugins as a source instead of the internal one to PrusaMMU.

chrissexton, 2025-02-28:
Interesting. I probably don't want to allow them full settings access, but I will check out those other options. Thank you.

closed 2025-02-28 by chrissexton