jukebox42/Octoprint-PrusaMMU #64: Core One: PrusaMMU unknown

Mercur13, 2025-08-09:
What model of Prusa printer are you using? Core One
What version of the Prusa MMU plugin are you running?  2025.1.27
What versions is your printer running? 6.3.4  what version is your MMU running?)  3.0.3+896
What behavior are you seeing?

MMU is displayed as "Unknown".

No Filament-Optionlist is displayed

Mercur13, 2025-08-10:
[serial.log](https://github.com/user-attachments/files/21704266/serial.log)

Thanks for your fast replay. I hope this is the file you want. 

If you want more, just ask.

Mercur13, 2025-08-11:
> Exactly what I was looking for thanks!
>
> Alright, if you are down for a little bit of testing, here's an experimental build. From what I can tell, the MMU support looks a lot like what they did for the MK4 so we may be able to get away with just adding your printer to list. So that's our first test. The changes so far are in #65 <https://github.com/jukebox42/Octoprint-PrusaMMU/pull/65> .
>
> You'll need to install this version of the plugin, you can do that by going into the plugin install menu and selecting from file and uploading this ZIP file. You should get 25.8.10alpha1 (it will tell you your version is out of date when you restart, ignore that warning)
>
> Octoprint-PrusaMmu.zip <https://github.com/user-attachments/files/21706750/Octoprint-PrusaMmu.zip>
>
> Stuff to test:
>
> *	Check and ensure you get the Idle message on the mmu in the nav.
> *	Go into the settings for the plugin, scroll down and check what version it detected. it should say COREONE
> *	Under settings, near the bottom there's an "advanced options" section, if you turn debug mode on you'll be able to see additional log info in your browsers devtools console. We'll need this info if anything doesn't work as expected
> *	Try a print and ensure you get the popup (with the Core you should get the popup every time)
> *	Try selecting the no override option and printing like normal
> *	On another print, try selecting a tool that does not match one you sliced with and ensure it picks the right tool.
>
> Thanks in advance!

Hello John,

 

good news, all works as expected !

 

*	the Idle message on the mmu in the nav. OK
*	check what version it detected. it should say COREONE  OK
*	log info in your browsers devtools console. “Feeding …” and so on  OK
*	Try a print and ensure you get the popup (with the Core you should get the popup every time)  OK
*	Try selecting the no override option and printing like normal  OK
*	On another print, try selecting a tool that does not match one you sliced with and ensure it picks the right tool.  OK

Any further wishes  to test ?

Jens

jukebox42, 2025-08-16:
Thanks! This was easier than expected. Appreciate the support. I've added you to the special thanks in the readme.

The latest release includes these changes above :)

closed 2025-08-16 by jukebox42